Morningstar-Engineering/MC8-MIDI-Controller #3: Add EHX HOG2 to MIDI Dictionary

geoffrey-eisenbarth, 2020-09-03:
Not sure if this is the proper channel for this, but I was wondering if Morningstar accepts "pull requests" or something similar in regards to adding pedals to the MIDI dictionary. The specification for the EHX HOG2 is listed in it's manual, and figured it might be nice to have it near by instead of having to tab over (and rotate the PDF). 

Great work on this controller guys, it's simply amazing.

tyzjames, 2020-09-03:
@geoffrey-eisenbarth 

We just opened our MIDI dictionary for community contribution. More info here: https://github.com/Morningstar-Engineering/openmidi

This is linked to the MIDI dictionary in the editor as well.

https://www.openmidi.com

geoffrey-eisenbarth, 2020-09-03:
Awesome, thanks for the swift response, I'll look into those resources!

closed 2020-09-03 by geoffrey-eisenbarth