Morningstar-Engineering/MC8-MIDI-Controller #4: Saving bank messages problem

psweetie, 2021-05-26:
In the 3.9.0 beta I was saving set expression messages in various banks such that the expression pedal worked for that pedal when the bank was entered. The message ended up being replicated across multiple banks and the names of the two banks got corrupted. When I went back to the release version the messages saved fine and no problem with the Bank names.

tyzjames, 2021-05-27:
thanks for the feedback. was this on the latest firmware? In one of the first few releases there was an issue with the save function from the editor that might have caused what you encountered. Do you have the filename of the firmware you used?

psweetie, 2021-05-27:
Hi James

File name is 2021-05-19_v2_MC8_Firmware_v3_9_0_Beta.hex

Regards
Philip

***@***.******@***.***>

psweetie, 2021-05-27:
Hi James

Just downloaded latest version and it doesn’t appear to do the same thing, it appears to act correctly. Will continue to use and report if it errors

Regards
Philip

***@***.******@***.***>

tyzjames, 2021-05-27:
Thanks for the feedback! Ok do let me know if you face the same issue again.

closed 2021-07-30 by psweetie