Morningstar-Engineering/MC8-MIDI-Controller #5: Chrome ties itself up

psweetie, 2021-07-02:
When using the 3.9.0 beta firmware on my MC*, every now and again the staging web editor stops working, wont reconnect and chrome has to be exited. There are then multiple Chrome processes left running (usually about 5 or 6) which have tyo be manually stopped in Task Manager before Chrome can be restarted and connection re-established.

Philip

tyzjames, 2021-07-27:
Hi Phillip, are you still facing problems with this in the latest release?

psweetie, 2021-07-27:
I have had it a couple of times but potentially more related to mc8 freezing with he global toggle piece?

Regards
Philip

***@***.******@***.***>

psweetie, 2021-07-30:
It has happened again with latest builds and firmware. Looking at Task Manager it shows 5 or 6 Chrome processes open and won't let you load Chrome until you stop these rogue processes. I can't reproduce exactly what leads to the situation though, just lots of edits with MC8 and other MIDI pedals then suddenly no comms with MC8 and Chrome hangs

tyzjames, 2021-08-02:
> It has happened again with latest builds and firmware. Looking at Task Manager it shows 5 or 6 Chrome processes open and won't let you load Chrome until you stop these rogue processes. I can't [...]

Thanks for reporting this. Do you have anything connected to the MIDI IN port in the MC8? What's the MIDI signal chain like?

psweetie, 2021-08-05:
i just have midi out connected to hx stomp in. then hx stomo out connected to source audio neuro hub in

Philip
***@***.***
+44 (7540) 664785

psweetie, 2021-09-03:
This does still happen with latest staging and firmware. Every now and again Chrome stops speaking to the MC8 and quitting Chrome still leaves some Chrome processes running, typically 5 or 6 instances, which have to be stopped before Chrome can be restarted and MC8 connected again. Can't find a pattern that causes it.

tyzjames, 2021-09-06:
thanks for the update. My main suspect there's some sort of MIDI loop going on. Are you on Mac or Windows? You can use this just to check the incoming MIDI messages, so when the chrome hangs, just take a look at the monitor to see if there is a huge chunk of midi messages streaming into the computer

psweetie, 2021-09-09:
On PC, Win10. Presumably you mean the Morningstar editor monitor feature. Will try that but not possible if chrome has stopped talking to the MC8 and obv can't connect any other MIDI tool either as they report MC8 connected elsewhere . . .

tyzjames, 2021-09-10:
I meant an external midi monitor software not related to chrome. On Mac, I use this: https://www.snoize.com/midimonitor/

On windows, you can use this to monitor incoming MIDI messages: http://www.midiox.com/

psweetie, 2021-09-10:
ok, have that installed and see no storms so far, i did turn off midi clock persist just in case, and notice that if you have no exp pedal plugged in there are a load of messages looping . . .

tyzjames, 2021-09-10:
What kind of messages?

psweetie, 2021-09-10:
I have an expression set up with CC99 on Msg 1 and CC100 on Msg 2. If the expression pedal is not plugged in it constatntly send out the default CC99 message. I do have a trs cable connecting the exp 4 input on the MC8 to an extarnal jack into which I normally plug the exp pedal.

psweetie, 2021-09-10:
stops as soon as the exp pedal plugged in to the external jack

closed 2021-11-01 by psweetie