Morningstar-Engineering/MC8-MIDI-Controller #6: Toggle Mode

psweetie, 2021-07-12:
Current Toggle Mode I believe is activated on a button Press action. Thus it is activated when the user Long Presses or Double Taps which gets in the way of separating those functions. You have to disable Toggle Mode and add a Toggle Preset command on Release to get that more separated functionality. Would it be better to have it run by default off a Release?

tyzjames, 2021-07-12:
Yes, the steps you specified are correct if you just want to toggle the preset with Release action. 
You'll need to add that toggle preset message with the selected action. We can't change it to be release by default because that will break the workflows for other users

closed 2021-07-12 by tyzjames

psweetie, 2021-07-12:
Thanks James, understood

Regards
Philip

***@***.******@***.***>