Morningstar-Engineering/MC8-MIDI-Controller #7: expression (scale) not working as I expected

psweetie, 2021-07-22:
I had an expression preset programmed to output from 0 to 127 over full range of movement. Looking at the MIDI monitor it did exactly that. I replaced the message with an Expression (scaled) message with a scale of 10 and a min of 0 and max of 127. However the output on the monitor never fell below @60. When applied to a volume block in my HX Stomp it therefore never turned it off. From the curves in the manual I would expect it still to go to zero at heel down, just to change the curve itself?

tyzjames, 2021-07-27:
Thanks for raising this. I thought it might be an issue with the sensitivity setting being low but I was able to get the proper range of values between 0-127 at all the different sensitivity setting. Can you check if your sensitivity is set to the highest, and that expression pedal is calibrated in the MC8?

psweetie, 2021-07-27:
Expression pedal is calibrated and shows right values in every other application on the mc8
Sensitivity setting is default 3 in the settings page and haven’t fiddled with that as yet?

Regards
Philip

***@***.******@***.***>

psweetie, 2021-07-27:
Sensititivity seems to make no difference. With no scale the expression pedal is fine and monitor show 0-127. With scale and scale factor set at at 10 it shows @40-127. As you reduce the scale the lower figure also gets lower but does not go to zero.

Have tried calibrating pedal several times and also makes no difference.

Going back to linear always resolves to 0-127

Regards
Philip

***@***.******@***.***>

tyzjames, 2021-09-06:
Thanks for reporting this. I just checked it out and saw the issue. Have fixed it and it will be in the next release

psweetie, 2021-09-07:
Looks like it now runs correctly form 0-127 James, thanks.
Presumably the % expression will remain the same but the actual MIDI CC message output will be scaled?

closed 2021-09-15 by psweetie