Morningstar-Engineering/MC8-MIDI-Controller #11: Bank settings not loading on entering bank

Classic07, 2021-08-20:
Bank settings set for on entry are not activating when entering the bank

tyzjames, 2021-08-20:
Can you share a screenshot of your preset settings in the editor for this? Are you referring to the Bank Jump message or the Bank Preset On Enter messages?

Classic07, 2021-08-20:
[image]

The issue is exhibited on bank jump rather than specifically entry. The bank jump command seems to result in the 'on entry' commands being ignored.

tyzjames, 2021-08-20:
check your Bank Jump messages - there are now additional settings there so you can select whether to execute bank presets on enter/exit when you bank jump. It might be turned off by default. We have not implemented a function to initialise all these values for this update yet

Classic07, 2021-08-20:
That's it thanks tyuzjames. Which setting do I need to enact destination presets on entering the new bank?

Classic07, 2021-08-20:
Don't worry - all sorted thank you!

closed 2021-08-20 by Classic07