Morningstar-Engineering/MC8-MIDI-Controller #13: Staging Editor not showing bank or preset names in the lists

psweetie, 2021-09-02:
Just shows Bank 1 - Bank 1, etc. and Preset A - 
The Banks and Presets are there but not shown on the list
I tried restoring from previous json file and issue still persists

Also in the drop downs when you try and change a CC number, say, the drop down list shows white text on white background and has done for a while. Only shows a number when the focus bar is on it!

psweetie, 2021-09-02:
just to be clear - running latest firmware as of today just in case, wasn't doing this before that update

tyzjames, 2021-09-02:
@psweetie I just pushed the update to the staging editor so there might be a lag. can you refresh? Also, I just re-uploaded the 2021-09-02 file

tyzjames, 2021-09-02:
Noted on the dropdown box issue - it seems to be only affecting Windows - will dig into it

psweetie, 2021-09-02:
note that if I save the preset its name comes back again. will try and refresh and upload firmware again James

psweetie, 2021-09-02:
latest staged version works again James, bit too eager!!

tyzjames, 2021-09-02:
I actually rolled back the update a few hours ago after uploading your file - as the controller seemed to hang on start up, but turns out its the bank preset Delay message that is delaying for 10 seconds when set to 1 second. Have fixed that issue as well.

closed 2021-09-03 by psweetie