Morningstar-Engineering/MC8-MIDI-Controller #15: What triggers the Toggle Mode On/Off

psweetie, 2021-09-06:
There was a problem with calling a Preset Engage from another Preset where it didn't appear to be taking care notice of the toggle position. That is ficed in latest update. I still have a problem though.

What I want is all the complicated programming to be setup in one Bank and one Preset and for it to be associated with a Global Toggle variable such that I will know when it is on/off. I typically use a Release message to do all the work. Also worth saying I have a long press action on these Presets so do not want the release actions triggered from a long press and vice versa. I then want another Preset in another Bank just to call Engage Preset with a Release for the Release functionality to work and similarly for the Long Press.

I can get the Release functionality to work if only Release messages are there and I use Toggle Mode On on the complicated Preset. If I add the Long Press functionality it seems to also trigger the Release functionality and switches the toggle position which is not wanted! If I substitute the Toggle M ode On for Off and use the Set Toggle command the remote call only turns the pedal on and never gets to the second state.

Clearly I can replicate all the programming so this is 1st world problem, but what I was trying to achieve was a "remote" switch that could call functionality of another switch and inherit the state(s). It may be this is too complicated and I should just replicate functionality each time . . . ?

psweetie, 2021-09-06:
As I said above I want to create a "remote" switch that calls another Preset in another Bank and replicates its visuals showing when the main Preet is on/off/ and also shifted if possible.

I attach my backup:
[Morningstar_MC8_All_Banks_Backup_File (26).txt](https://github.com/Morningstar-Engineering/MC8-MIDI-Controller/files/7118044/Morningstar_MC8_All_Banks_Backup_File.26.txt)

Bank 10 is a Test 1 bank and has Preset A set with Toggle Mode Off. It is activated by Preset D which is supposed to mimic the visual aspects of Preset A without replicating all the programming. It pretty much works, albeit you do have to re-programme the shift stuff into the remote preset. I can even get Preset D to use the same  Global Preset Toggle Group as Preset A and therefore blink in sync etc. If I try to replicate that functionality in Bank 11, which I have done on Preset C, it does not work and it seems to be the Global Toggle Group that is getting in the way this time. Make it Local and the basic functionality works, albeit it now will not reflect the state of the original.

Again this may be  unintended behaviour and may be because I am trying to do something not envisaged in current use cases. Should I be doing it a different way? Use case is, as you see in the attached backup, I have Banks setup for Drive, Mod, Delay, etc. And want a Stomp Box Bank to access some of those Presets defined in their own banks and thought there should be an easier way to get it rather than just replicating the functionality of ebery button?

Philip

psweetie, 2021-09-15:
Just checked 09_15 version and the behaviour has not changed . . .

tyzjames, 2021-09-15:
I've not looked into this yet, will look into it tmr and understand the issue

psweetie, 2021-09-15:
Thanks James
By the way I would be more than happy to test a Pro MC8 when we get more chips around! 😉

Regards
Philip

***@***.******@***.***>

psweetie, 2021-09-17:
I have been experimenting again to see if I was doing anything obviously wrong or stupid. What I am trying to create is effectively a remote button that can call all the functionality of a Preset elsewhere on the MC8 without having to reprogramme all the code. I can then setup banks for delays, reverbs etc. and a stomp box bank which has highlights of the others . . .

In Bank10PresetA I have Release and a Long Press and Shift actions programmed. It only works with the Shift functionality correctly if Toggle Mode is turned of and Toggle Preset is programmed on the Release event. Button works as planned. It turns on and off the pedal on on subsequent Releases and when on, a Long Press operates the shift and adds a boost. Another Long Press deactivates the Shift but leaves the pedal still on. Pressing preset in that state then turns pedal off. My thought is that this should all work with the Toggle Mode On and the toggle should not be triggered by the Long Presses but just though a Release action, butI can get around that.

I have then set up Bank10PresetD to just Engage Preset A on same Bank. It uses the same global toggle as Bank10PresetA so it will blink in sync. I do not have Toggle Mode on, just engage Preset with a Release action every time the button is pressed. This works as a remote. It turns pedal on and off and it blinks in the right places.

I also have Bank10PresetC setup with same Release event but also with the Long Press and Shift functionality programmed. I had to programme some of that in the switch to make sure the visual aspect works on the remote switch. As with the above this works fine. Once again I did not use Toggle Mode but shared the same global preset.

If I transplant these remote buttons to Bank11 and still call the Bank10PresetA I get the problems, they do not work as they do in the same bank. Typically they will turn on the pedal and not turn it off, so somewhere between the Banks the status of the presets (Pos 1 or Pos2) gets confused, even though a Global Preset is being used which should keep its value.

Philip

[Morningstar_MC8_All_Banks_Backup_File (4).txt](https://github.com/Morningstar-Engineering/MC8-MIDI-Controller/files/7184451/Morningstar_MC8_All_Banks_Backup_File.4.txt)

psweetie, 2021-10-06:
> Works ok in same bank but engage preset with release used across banks still does not trigger a change in state whether using local or global toggle presets

Hi James

  *   Backup file attached
  *   Bank 10 Preset A is the one I am trying to activate remotely. It has a Release action to turn on and off a pedal
  *   In Bank 10
     *   Preset B sends an Engage Preset with Release action to Preset A and used the Toggle Group
     *   Preset C sends an Engage Preset with Release action to Preset A and used the local Toggle
     *   Both of these work as expected
  *   In Bank 11
     *   Preset A sends an Engage Preset with Release action to Bank 10 Preset A and used the Toggle Group
     *   Preset B sends an Engage Preset with Release action to Bank 10 Preset A and used the local Toggle
     *   Neither of these work as expected. They turn on the pedal BUT never turn it off
     *   Note that the status never changes yet the displays does alternate between flashing and static

Regards
Philip

***@***.******@***.***>

I fixed some bugs related to using Engage Preset to engage Set Toggle in current or other banks with presets linked to global toggle states. I haven’t used your file yet, but it might solve your issue

{"schemaVersion":1,"dumpType":"allBanks","deviceModel":4,"downloadDate":"2021-10-06T12:16:11.308Z","hash":1877554965,"data":{"bankArray":[{"bankNumber":0,"bankName":"Home","bankClearToggle":false,"bankMsgArray":[{"msgNum":0,"data1":3,"data2":0,"data3":1,"data4":0,"channel":1,"type":7,"action":1,"toggle":2,"msgInfo":""},{"msgNum":1,"data1":3,"data2":0,"data3":1,"data4":0,"channel":1,"type":7,"action":2,"toggle":2,"msgInfo":""},{"msgNum":2,"data1":32,"data2":0,"data3":0,"data4":0,"channel":1,"type":6,"action":2,"toggle":2,"msgInfo":""},{"msgNum":3,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":4,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":5,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":6,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":7,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":8,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":9,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":10,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":11,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":12,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":13,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":14,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""},{"msgNum":15,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":0,"msgInfo":""}],"presetArray":[{"presetNum":0,"bankNum":0,"isExp":false,"shortName":"Drive>","toggleName":"","longName":"","toToggle":false,"toBlink":false,"toMsgScroll":false,"toggleGroup":0,"msgArray":[{"msgNum":0,"data1":1,"data2":0,"data3":6,"data4":0,"channel":1,"type":13,"action":2,"toggle":2,"msgInfo":""},{"msgNum":1,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":2,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":3,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":4,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":5,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":6,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":7,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":8,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":9,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":10,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":11,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":12,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":13,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":14,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":15,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""}]},{"presetNum":1,"bankNum":0,"isExp":false,"shortName":"Mod>","toggleName":"","longName":"","toToggle":false,"toBlink":false,"toMsgScroll":false,"toggleGroup":0,"msgArray":[{"msgNum":0,"data1":2,"data2":0,"data3":6,"data4":0,"channel":1,"type":13,"action":2,"toggle":2,"msgInfo":""},{"msgNum":1,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":2,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":3,"data [... cut ...] :5,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":6,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":7,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":8,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,"action":0,"toggle":2,"msgInfo":""},{"msgNum":9,"data1":0,"data2":0,"data3":0,"data4":0,"channel":1,"type":0,