Morningstar-Engineering/MC8-MIDI-Controller #17: Toggle Mode On and Long Press

psweetie, 2021-09-16:
If you want to have both Long Press and a Press commands operate separately on the same Preset advice is to use Release rather than Press then the two do not interact. However, if you turn on Toggle Mode it is triggered by both the Release action and also by the Long Press action. Is this intended behaviour? In which case you have to turn off the Toggle Mode and use a Toggle Prest command to determine exactly when the toggle state changes.

tyzjames, 2021-09-17:
Yes, if Toggle Mode is on, it will be triggered by any action.  If you want a specific action to toggle the preset, you should use the Toggle Preset action.

psweetie, 2021-09-17:
Thanks James, that makes sense

Regards
Philip

***@***.******@***.***>

closed 2021-09-17 by psweetie