Morningstar-Engineering/MC8-MIDI-Controller #18: Release 3.9.0 29th Sept Bricks MC8

Classic07, 2021-09-29:
The latest release causes the MC8 to brick and become unresponsive to button press and does not reboot. Once powered off and back it only shows blocks on screen. However, the device _is_ responsive behind the scenes and can accept the previous release version from 10th September and the device reboots and works correctly again.

My PCB is the blue version and hasn't been updated to the latest/updated version yet.

tyzjames, 2021-09-29:
Thanks for the report. Nothing major has been changed besides adding some new features with regards to message types, so it is unlikely that that will happen. I'll test the upload on another MC8, but in the meantime, can you confirm that you uploaded the correct firmware meant for the MC8 (not MC6 or MC3)?

Classic07, 2021-09-29:
Discovered what's causing it and you are correct, the firmware is for the MC6 however that's the firmware download available on the **MC8** page. Note the URL and text at the top of the page:

[image]

[image]

tyzjames, 2021-09-29:
my bad! I uploaded the wrong files. I have fixed it - do check and let me know if you continue to face any issues.

closed 2021-09-29 by tyzjames

Classic07, 2021-09-29:
The latest MC8 files are now available and it's working as expected now, thank you. 

[image]
!