Morningstar-Engineering/MC8-MIDI-Controller #19: Activity Monitor

psweetie, 2021-10-04:
Really liking the ability to see what is actually being triggered. Useful to debug your messages. Would be very useful to see the state of the toggle / toggle group (incl shift) as part of the readout . . .

tyzjames, 2021-10-04:
ah yes, toggle position would definitely be useful. Will look into adding that, thanks!

tyzjames, 2021-10-06:
done

closed 2021-10-06 by tyzjames

psweetie, 2021-10-06:
Looking good, what does it mean when it shows Position: Both, I thought it was a binary option of Pos 1 or Pos 2?

Regards
Philip

***@***.******@***.***>

psweetie, 2021-10-07:
> It was initially showing the message toggle setting. But I have changed it to the preset toggle state in today's update

That is great

Regards
Philip

***@***.******@***.***>