Morningstar-Engineering/MC8-MIDI-Controller #22: Storm of SysEx messages

psweetie, 2021-10-29:
In the latest release, when I issue a MIDI Clock Tap message I get a storm of SysEx messages tyhat just keep on going. Have tried with MIDI persist on and off and makes no difference.

psweetie, 2021-10-29:
backup file
storm created when in Bank1 and using Tap functionality on Preset E

[Morningstar_MC8_Backup_File_20211029.txt](https://github.com/Morningstar-Engineering/MC8-MIDI-Controller/files/7444137/Morningstar_MC8_Backup_File_20211029.txt)
 
Philip

tyzjames, 2021-10-30:
The SysEx messages is actually incoming MIDI clock messages - I think I accidentally removed the filter. I have added it (the filter) back in the editor - not related to the MC8 firmware.

closed 2021-10-30 by tyzjames

psweetie, 2021-10-30:
Aha, thanks James, certainly Janet seen them before andi an returning incoming MIDI clock in system settings

Philip
***@***.***
+44 (7540) 664785