Morningstar-Engineering/MC8-MIDI-Controller #23: MC8 on Firmware 3.9.3 Crashing

Classic07, 2021-12-05:
I played a gig tonight with the normally stable MC8 which crashed three times during a 3 hour timeframe, which posed a few challenges. The MC8 has recently been updated to the latest firmware 3.9.3 as of the 12th November and has until now been fine.

I was previously running the beta version which for my needs was completely stable.

Any suggestions or recommendations? Did I need to do anything specific as part of the update to the live version that I may have missed?

tyzjames, 2021-12-05:
there is a bug when updating from v3.8 to v3.9, which requires the data to be backed up in v3.8, and then reloaded into v3.9 after the update. this issue affects the bank presets. So im guessing there an illegible value after the update that is causing the controller to hang. 

Are you using bank presets in your controller? Can you just check that all the bank presets are of the correct values, if you do not have the data dump file to load?

Classic07, 2021-12-05:
If by bank presets am I using a song list, yes I am. I'll try the data load trick and also check the bank preset values.

tyzjames, 2021-12-05:
ok, let me know. So far we have had a couple of these freeze issues from other users and it was due to the bank presets. But if you have already edited the bank presets in v3.9.3 then there's not point doing the data dump. just check your bank preset values to see if there's anything illegible

Classic07, 2021-12-05:
Trying to restore controller presets and got a 'Failure-2' message twice then it appeared to be successful on the third attempt. Is this something I should be concerned with?

tyzjames, 2021-12-07:
Nothing to be concerned with - sometimes the communication between software and device is flaky so there might be missing bytes. There are retry mechanisms in place but sometimes it might fail. It's on us to enhance this.

Classic07, 2021-12-07:
I have rehearsal tonight and will validate if the reload and/or bank presets have been resolved.

Classic07, 2021-12-07:
I used the controller for almost 3 hours tonight without any issues. I have a gig Saturday night so will close this issue and review after Saturday's gig.

closed 2021-12-07 by Classic07