Morningstar-Engineering/MC8-MIDI-Controller #25: Using a switch in one bank to remote preset in another bank broken

psweetie, 2022-01-07:
I have a "stomp" bank which basically calls "engage preset" with release set for presets in other baks. Thgis prevents me having to replicate complicated MIDI command stacks. Working finr in 3.9.6 but broken in 3.9.7 when loaded this morning. Back to 3.9.6 and all works again . . .

Philip

tyzjames, 2022-01-08:
Thanks for reporting this. Yes I was experiencing intermittent behaviour when using engage preset. I have fixed the issue in today's release. If you do manage to give it a go, let me know if the issue is resolved for you

psweetie, 2022-01-08:
Thanks James, will try this later today

Philip
***@***.***
+44 (7540) 664785

psweetie, 2022-01-08:
Hi James
Will test more thoroughly tomorrow, but looks like remote operations are working again.

Regards
Philip

***@***.******@***.***>

psweetie, 2022-01-09:
version 0108 fixes the problem for me

closed 2022-01-09 by psweetie