Morningstar-Engineering/MC8-MIDI-Controller #26: Engage Preset seems to have no action within same Bank

psweetie, 2022-02-08:
If I use Engage Preset from within the same Bank there seems to be no action at all, jist ignores it. If I use the individual PC/CC commands everything works but issuing an Engage Preset with Release action does not trigger those actions. Works fine across Banks.

tyzjames, 2022-02-09:
Thanks! this is a known issue and we have fixed it - waiting to release in next minor update. looking through a few more bug first.  Attaching the latest build here
[Uploading 2022-02-08_MC8_Firmware_v_3_9_7.hex.zip…]()
.

psweetie, 2022-02-09:
Thanks James, that link doesn’t seem to work?

Regards
Philip

***@***.******@***.***>

psweetie, 2022-02-09:
Got it, had to be within GitHub

Regards
Philip

***@***.******@***.***>

psweetie, 2022-02-09:
that works, thanks James

closed 2022-02-09 by psweetie